Telecominfraproject/OpenCellular #47: Feature request: GBC-V2 Cavium support for DVT

swateeshrivastava, 2018-10-01:
### *Product:* (OC-SDR, OC-LTE, OC-LTE, OC-Power, others)
OC-SDR

### *Component:* (PSU, GBC, BBU, FEM, LED, ME)
GBC
   
### *Category:* (Hardware, Software/stack, Firmware/bootloader, Documentation)
Software/stack
   
**Describe the solution you'd like**
Description : Software validation support for all GBCV-2 peripherals.

Excel sheet is attached for all the software requirements.
[Test_plan_for-GBCV2.xlsx]
(https://github.com/Telecominfraproject/OpenCellular/files/2433591/Test_plan_for-GBCV2.xlsx)

**Is your feature request related to a problem? Please describe.**

arulbose, 2018-11-12:
Attached GBC DVT software support doc.
[GBC_DVT.zip](https://github.com/Telecominfraproject/OpenCellular/files/2571819/GBC_DVT.zip)

arulbose, 2018-11-14:
[GBC_DVT.docx](https://github.com/Telecominfraproject/OpenCellular/files/2581999/GBC_DVT.docx)
Updated host application testing to the doc

arulbose, 2018-12-06:
Closing the ticket as DVT SW support almost completed and no more new request from HW team.

closed 2018-12-06 by arulbose