Telecominfraproject/OpenCellular #174: BB-LTE package release for testing

arulbose, 2018-10-26:
### *Product:* (OC-SDR, OC-LTE, OC-LTE, OC-Power, All, N/A, others)
   OC-LTE
### *Component:* (PSU, GBC, BBU, FEM, LED, ME)
BBU
### *Revision:* (A, B, C ...)
Rev C
### *Category: (schematic, layout, BoM, gerber, compliance, testing, documentation, etc.)
Testing
### *Clearly explain the issue in detail, with snippets of schematic (or ref to the exact page), layout, etc, as needed*
Currently changes in Rev C board is managed with u-boot environment variables and tester has to save the environment variables every time after flashing the bootloader hence to make the environment variable changes permanent there is a need add the env in the u-boot build. Also the telnetd deamon does not start automatically in the rootfs. A minor change in the startup script to be added. Since BB rootfs resided in ramdisk there is a need to provide rootfs build too. This release will have bootloder and rootfs which will use the existing tip kernel without any changes.

arulbose, 2018-10-31:
@ramarao-bitra please test the release binary and let us know the results. Thanks

ramarao-bitra, 2018-11-02:
@arulbose Please share me the uboot bin file

arulbose, 2018-11-02:
[uboot_added_env.tar.gz](https://github.com/Telecominfraproject/OpenCellular/files/2541347/uboot_added_env.tar.gz)
@ramarao-bitra  PFA.

arulbose, 2018-11-12:
Tested and working. It has all the new environment variables added. Attached modified u-boot TIP board config file

[octeon_tip.zip](https://github.com/Telecominfraproject/OpenCellular/files/2571725/octeon_tip.zip)

closed 2018-11-12 by arulbose